fatihak/InkyPi #237: critical inkypi dead ? Help needed

Big-Brummer, 2025-07-28:
I take a copy of the calender display to my PC browser 
http://192.168.199.92/static/images/current_image.png
It is great  
BUT the Inkypi did not perform any update after this simple request.
It is still responding via ssh and the web interface reacts also
BUT the display is frozen and does not do any updates at all.
even sudo reboot or power down and up did not fix this issue.
any idea ?

fatihak, 2025-07-29:
So you're still able to access the web ui but the display isn't updating? Can you check the logs for any errors through the "Download Logs" button in the settings page or by running this command:
`journalctl -u inkypi -n 100
`
Can you also provide details on which Raspberry Pi Model and Display Model/Size you have?

Big-Brummer, 2025-07-30:
Thank you very much for your help.
I am using zerro 2W and the Light version without desktop via ssh.
Unfortunately, the logs were lost due to power down and up. However, the display still has not been updated.
Only an update of the software according to your instructions was successful. Although Inky was up to date, approximately 4 files were replaced and now the calendar is working again.
However, I don't dare enter the command “http://192.168.199.92/static/images/current_image.png” into my browser again because I don't know if it's permissible and if Inky will crash again. A reset fixed error is really critical.

phattmatt, 2025-07-31:
Hi @Big-Brummer,

Looks like you resolved the problem; can you consider closing this issue? Thanks.

Big-Brummer, 2025-07-31:
I will close the issue because the update did solve it.
But allow me one last question.
Will it be ok to call that page via my PC Browser ?
How many client will inky accept ?
“[http://192.168.199.92/static/images/current_image.png”](http://192.168.199.92/static/images/current_image.png%E2%80%9D)

closed 2025-07-31 by Big-Brummer