fatihak/InkyPi #245: Moon phases disappear from weather plugin

ian009, 2025-08-02:
The moon phase data disappears from the display when the playlist weather plugin is modified 

The moon phase should not disappear if it is still selected

Steps to reproduce the behavior:
Configure the weather plugin with moon phase selected
Add it to the playlist & run
Confirm that moon phase is displayed
Modify any parameter C to F or how may days to show forecast
Do not uncheck the Moon phase
After saving & when the display refreshes the moon phases are not displayed

Please see screenshots

[image]
[image]

phattmatt, 2025-08-02:
The above change fixes the issue for me. I'll raise a small PR when I get a chance, unless someone beats me to it.

Leberkasbepi, 2025-08-02:
Works but 0%? 😅

[image]

phattmatt, 2025-08-02:
> Works but 0%? 😅

I don't know what is going on there. The fix is working for me, and I can't see how the change would affect the API response, or how the data is rendered.

For each weather plugin instance you will need to edit settings, untick and then re-tick the Moon Phase checkbox, and then click save. This will correctly set the configured value to 'true'.

PR: https://github.com/fatihak/InkyPi/pull/247