fatihak/InkyPi #248: Use DOW, MONTH DAY(st,nd,rd) instead of DOW, MONTH DAY (i.e. Saturday, August 2nd)

TonyTTTT, 2025-08-03:
Add function:
def get_ordinal(n):
    if 10 <= n % 100 <= 20:
        suffix = 'th'
    else:
        suffix = {1: 'st', 2: 'nd', 3: 'rd'}.get(n % 10, 'th')
    return f"{n}{suffix}"

and in def parse_weather_data and in def parse_open_meter_data
Change 
    "current_date": dt.strftime("%A, %B %d"),
 to     
     "current_date": f"{dt.strftime('%A, %B')} {get_ordinal(dt.day)}",

fatihak, 2025-08-24:
I don't think this is necessary, the existing date format is clear and commonly used across different weather services/dashboards:

[image]

[image]

[image]

[image]

closed 2025-08-24 by fatihak

TonyTTTT, 2025-08-27:
ok, but you should format without leading 0's, so that its August 2, and not August 02