fatihak/InkyPi #253: Feature request: Integration with Tempest Weather

moorew, 2025-08-06:
Hey - I love your work, and have the weather display up and running on my desk. I would howver love to get it working with my personal Tempest Weather station (and would be willing to pay you, either as a sponsor or a one off donation!)

1) API endpoint and authentication

    - URL
`https://swd.weatherflow.com/swd/rest/observations/station/{station_id}?token={api_token}
`

    - Auth
        - Simple query parameter ?token=
        - No OAuth or extra headers

2) Required settings

- Provider name: “Tempest” in the Weather dropdown
- API token (string)
- Station ID (integer)
- Latitude/Longitude (floats) for sunrise/sunset, icon timing
- Units (“metric” or “imperial”)

3) JSON payload keys
From the `obs[0]` object you’ll get:

```
{
  "air_temperature":       number,  // °C
  "relative_humidity":     number,  // %
  "wind_avg":              number,  // m/s
  "wind_direction":        number,  // degrees
  "sea_level_pressure":    number,  // mb
  "precip_accum_local_day":number,  // mm
  "strike_count":          number   // lightning strikes
}
```
4) Suggested implementation steps
- In `src/plugins/weather/…`, add “Tempest” to the provider list

- In the plugin blueprint, expose tempest_token and tempest_station_id fields

- In fetch_weather (or generate_image), branch on weather_provider == "Tempest" and:

```
url  = f"https://swd.weatherflow.com/swd/rest/observations/station/{station}?token={token}"
resp = requests.get(url, timeout=10).json()
obs  = resp["obs"][0]
return {
  "temp":     obs["air_temperature"],
  "humidity": obs["relative_humidity"],
  "wind":     obs["wind_avg"],
  "pressure": obs["sea_level_pressure"],
  "rain":     obs["precip_accum_local_day"],
  "strikes":  obs["strike_count"]
}
```
Respect the existing latitude/longitude and units logic for icons and formatting

Links:

Tempest API docs: https://weatherflow.github.io/Tempest/api/

Example station list endpoint:

`https://swd.weatherflow.com/swd/rest/stations?token={api_token}`

Happy to provide sample code or test credentials if you’d like. Thanks for considering this—Tempest support would be a big win for home-weather enthusiasts like me and I tried to vibe code it myself, but the machine still can't code like humans ;)!

I tried adding the below to `device.json` but it didn't work. I'm sure there is far more involved. 

`{
  "playlists": { "Default": ["weather"] },
  "weather": {
    "weather_provider":"tempest",
    "tempest_token":"86aaa490-0b91-471b-81d3-3f2618b3151a",
    "tempest_station_id":159913,
    "latitude":43.14279,
    "longitude":-80.2707,
    "units":"metric"
  }
}`

I work for a tech company (Tailscale) but in communications, I always wanted to learn to code, but never did, so now I tinker and play as a pretend devops at home LOL.

As an aside, I use Tailscale to make the frame available anywhere in the world - not just through .local. mine is accesible through my tailnet - would be a cool thing to add in instructions for users to do that too - Tailscale has an amazing free plan that would suit most folks for life. 

Cheers,
Will

moorew, 2025-08-12:
@phattmatt I see you added this as an enhancement... does that mean you'd be interested? I can give you my email if you want to chat about it :) Thanks for looking!

phattmatt, 2025-08-12:
Hi @moorew,

I added the enhancement tag to categorise the type of issue/request to help triage new issues. Bug fixes are likely to be prioritised.

As you can see there are a LOT of enhancement requests.

I am unlikely to take this request on, but by putting the enhancement tag on it may encourage someone else to give it a go.

By the way, I notice you're using Tail Scale to access the InkyPi away from home, is that right?

moorew, 2025-08-12:
> By the way, I notice you're using Tail Scale to access the InkyPi away from home, is that right?

Yep - I work for Tailscale, so it seemed rude not to :P 

Thanks and totally understand that this isn't a priority. Maybe I will figure it out myself one day - if so I'll be sure to share!

kalidatuna, 2026-09-11:
I put together a working Tempest proof here: https://github.com/kalidatuna/InkyPi/tree/proof/tempest-weather-integration. It uses Tempest for current station readings and Open-Meteo for forecast/hourly data, with 8/8 Tempest tests and 33/33 repo tests passing. If you're still interested, I can finish review tweaks and live-validate it for $20 fixed. I'm 17 in Georgia, so payment would need to be international bank transfer to Georgia or another legal direct method. I did not use the token posted in this issue, so please rotate that token before any live test.

parthchilwerwar, 2026-09-12:
Hi @moorew — I looked through the Tempest proposal in #768 and found a separate issue in the shared forecast code: imperial visibility converts metres as if they were feet, so 10 km shows as 1.9 miles. I've submitted the correction in #769, with six regression cases and all 31 repository tests passing. That patch is available independently of any paid work.

If you still need help getting Tempest working, I can offer a $10 fixed integration check: up to one hour reviewing the existing branch against sanitized station data, one reproducible parser/configuration correction if it fits that hour, and setup instructions with the test results. If a larger problem turns up, I would report it and agree a new scope before further work. You would run the final check on your own Pi; I don't have the display hardware and haven't accessed your station. The implementation already in #768 remains credited to its author.

Would that small scope be useful, and can you pay by PayPal after you accept it? If so, please confirm the scope and review/payment timing before I start that work. Keep tokens and payment details out of this thread.

Parth

moorew, 2026-09-14:
@kalidatuna and @parthchilwerwar thank you both for looking into this. I apologise for not updating the issue - I totally forgot about it tbh!

While I appreciate your efforts, I no longer have need of this, as I went down another path and built my own. You can see it at https://github.com/moorew/tempest-inky

I'll mark this issue as closed.

closed 2026-09-14 by moorew