fatihak/InkyPi #254: Colors washed out on waveshare epd4in0e + rotated image

hungry-somnambulist, 2025-08-06:
**Describe the bug**
I installed the epdconfig and  Colors washed out on waveshare epd4in0e

**Expected behavior**
I installed the epdconfig and epd4in0e.py files after copying InkyPi but the colors are unsaturated and washed out, the screen is also 90 degrees off.  I adjusted the H/W in the json and the .py file match the screen for the rotation but the colors i cant seem to address. 

**To Reproduce**
Steps to reproduce the behavior:
1. load any clock plug or wiki phot of the day the colors are washed out or incorrect. 

**Logs**
let me know what files or logs i can provide. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Using a rasberry pi 2 zero via hat connection

phattmatt, 2025-08-07:
Hi @hungry-somnambulist,

There is an existing bug issue open for the screen orientation, which seems to affect some screens (which default to portrait, rather than landscape): https://github.com/fatihak/InkyPi/issues/200.

There is an open issue about Waveshare displays that may be related to what you are experiencing: https://github.com/fatihak/InkyPi/issues/167.

Have you tried running any of the Waveshare examples, are they exhibiting the same display issues?

If so, the issue is likely to do with the Waveshare display and/or the Waveshare 'driver' code.

phattmatt, 2025-08-08:
For reference here are some photos of the clocks on my Inky Impression screen with the default colours:

[image]

[image]

And one of a clock with a pure blue (R:0, G:0, B:255):

[image]

closed 2025-08-26 by phattmatt