fatihak/InkyPi #257: Plugins relying on 'chromium-headless-shell' sometimes fail with "Bus error"

phattmatt, 2025-08-10:
> The error I'm getting with chromium-headless-shell in the new build (Bus error) looks like memory error.  
> The screenshot is loaded _sometimes_, but usually crashes.
> log attached.
> 
> [inkypi_20250810-072528.log](https://github.com/user-attachments/files/21705177/inkypi_20250810-072528.log) 

 _Originally posted by @TonyTTTT in [#159](https://github.com/fatihak/InkyPi/issues/159#issuecomment-3172579353)_

I've also seen this error, specifically happens on the Weather Plugin, weirdly the Calendar Plug in (both rely on chromium-headless-shell) still seems to be fine.

I suspect it MAY be related to the recent Chromium release (2025-08-06).

Downgrading to a previous release of Chromium fixes this for me.

phattmatt, 2025-08-12:
I've raised a small PR to temporarily pin the version of `chromium-common` and `chromium-headless-shell` packages to version '138.0.7204.183-1~deb12u1'

PR: https://github.com/fatihak/InkyPi/pull/265

This should be reversed when a proper fix is identified.

TonyTTTT, 2025-08-12:
I did the same, but with:
`sudo apt-mark hold chromium-common chromium-headless-shell`
Verify with:
`apt-mark showhold`
To later allow upgrades:
`sudo apt-mark unhold chromium-common chromium-headless-shell`

And to remind me when ssh in, I added this to .bashrc

```
holds=$(apt-mark showhold)
if [[ -n "$holds" ]]; then
        echo -e "\e[91mPackages are on hold:\033[0m"
        echo "$holds"
        echo -e "\e[92mAllow Package Upgrade with:\033[0m\n\tsudo apt-mark unhold chromium-common chromium-headless-shell"
fi
```

phattmatt, 2025-08-15:
I've posted on the official Raspberry Pi Forum to ask for help:

https://forums.raspberrypi.com/viewtopic.php?t=390969

TonyTTTT, 2025-08-16:
Error! An error occurred: Failed to take screenshot, please check logs.

```
Aug 16 17:00:11 inky-pi inkypi[596]: /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/inky/__init__.py:18: UserWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html. The pkg_resources package is slated for removal as early as 2025-11-30. Refrain from using this package or pin to Setuptools<81.
Aug 16 17:00:11 inky-pi inkypi[596]:   from pkg_resources import declare_namespace
Aug 16 17:00:12 inky-pi inkypi[596]: 17:00:12 - INFO - __main__ - Starting web server
Aug 16 17:00:16 inky-pi inkypi[596]: 17:00:16 - INFO - refresh_task - Starting refresh task
Aug 16 17:00:16 inky-pi inkypi[596]: 17:00:16 - INFO - waitress - Serving on http://0.0.0.0:80
Aug 16 17:00:26 inky-pi inkypi[596]: 17:00:26 - INFO - refresh_task - Manual update requested
Aug 16 17:00:26 inky-pi inkypi[596]: 17:00:26 - INFO - refresh_task - Refreshing plugin instance. | plugin_instance: 'Weather'
Aug 16 17:00:48 inky-pi inkypi[596]: 17:00:48 - INFO - plugins.weather.weather - Using location timezone for OpenWeatherMap data.
Aug 16 17:00:48 inky-pi inkypi[596]: 17:00:48 - INFO - plugins.weather.weather - Using timezone from weather data: America/New_York
Aug 16 17:01:04 inky-pi inkypi[596]: 17:01:04 - ERROR - utils.image_utils - Failed to take screenshot:
Aug 16 17:01:04 inky-pi inkypi[596]: 17:01:04 - ERROR - utils.image_utils - [0816/170055.558788:WARNING:media/audio/linux/audio_manager_linux.cc:54] Falling back to ALSA for audio output. PulseAudio is not available or could not be initialized.
Aug 16 17:01:04 inky-pi inkypi[596]: Bus error
Aug 16 17:01:04 inky-pi inkypi[596]: 17:01:04 - ERROR - refresh_task - Exception during refresh
Aug 16 17:01:04 inky-pi inkypi[596]: Traceback (most recent call last):
Aug 16 17:01:04 inky-pi inkypi[596]:   File "/home/pi/InkyPi/src/refresh_task.py", line 114, in _run
Aug 16 17:01:04 inky-pi inkypi[596]:     image = refresh_action.execute(plugin, self.device_config, current_dt)
Aug 16 17:01:04 inky-pi inkypi[596]:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Aug 16 17:01:04 inky-pi inkypi[596]:   File "/home/pi/InkyPi/src/refresh_task.py", line 279, in execute
Aug 16 17:01:04 inky-pi inkypi[596]:     image = plugin.generate_image(self.plugin_instance.settings, device_config)
Aug 16 17:01:04 inky-pi inkypi[596]:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Aug 16 17:01:04 inky-pi inkypi[596]:   File "/home/pi/InkyPi/src/plugins/weather/weather.py", line 122, in generate_image
Aug 16 17:01:04 inky-pi inkypi[596]:     raise RuntimeError("Failed to take screenshot, please check logs.")
Aug 16 17:01:04 inky-pi inkypi[596]: RuntimeError: Failed to take screenshot, please check logs.

```

TonyTTTT, 2025-08-17:
Might as well release it since the BUS error is already in the current version (needing a downgradie to chromium-headless-shell).

JZNine, 2025-08-24:
I ran an update today, and got this: 
Installing system dependencies...
E: Can't find a source to download version '1:139.0.7258.127-1~deb12u1+rpt1' of 'chromium-headless-shell:arm64'
E: Can't find a source to download version '1:139.0.7258.127-1~deb12u1+rpt1' of 'chromium-common:arm64'
E: Can't find a source to download version '1:139.0.7258.127-1~deb12u1+rpt1' of 'chromium-sandbox:arm64'

phattmatt, 2025-08-24:
> I ran an update today, and got this:
>
> ```
> Installing system dependencies...
>
> E: Can't find a source to download version '1:139.0.7258.127-1~deb12u1+rpt1' of 'chromium-headless-shell:arm64'
> E: Can't [...]

Run:

```
sudo apt update
sudo apt upgrade -y
```

JZNine, 2025-08-24:
Run, thanks and fixed. I was putting off running a full apt upgrade as it broke it last time, and was not sure chromium had been fixed since, and I had done the downgrade temp fix.