fatihak/InkyPi #263: Formatting issue in vertical mode for weather plugin

Leberkasbepi, 2025-08-12:
If you use InkyPi in vertical mode, then the formatting for the air quality does not fit. The unit of measure and the digits overlap.

[image]

phattmatt, 2025-08-12:
I've just checked on my InkyPi (Inky Impression 7.3" 2025 with 800x600) and I'm not seeing any issues:

<details><summary>Vertical Weather Image</summary>
<p>
[image]
</p>
</details> 

I note that you have your Weather in German, which is currently not supported natively by InkyPi. 

I assume you have edited the template files in the weather plugin folder?

It is probably this modification that has resulted in the template not formatting the screen well; you may need to tweak the template more to accommodate the differences in text.

phattmatt, 2025-08-12:
Ah, I've just checked using Open-Meteo (instead of OpenWeatherMap) and it appears that Open-Meteo returns a slightly different value resulting in a bit of a squeeze:

<details><summary>Open-Meteo Image</summary>
<p>
[image]
</p>
</details>

phattmatt, 2025-08-12:
Looks like the issue is the longer label used by the Open-Meteo code:

https://github.com/fatihak/InkyPi/blob/eb600591f7ffb032bed7c58a9dde255cb19b7c7e/src/plugins/weather/weather.py#L556-L559

Compared to the OpenWeatherMap code:

https://github.com/fatihak/InkyPi/blob/eb600591f7ffb032bed7c58a9dde255cb19b7c7e/src/plugins/weather/weather.py#L412-L417

This is due to the differences in how each weather service returns the air quality data.

OpenWeatherMap returns an Index:

https://openweathermap.org/api/air-pollution

Open-Meteo returns a particulate matter measurement:

https://open-meteo.com/en/docs/air-quality-api

fatihak, 2025-08-17:
I just updated the Open Meteo logic to pull & display the air quality index to be consistent with the Open Weather Map dashboard. Feel free to pull the latest changes and give it a try!

closed 2025-08-17 by fatihak