fatihak/InkyPi #272: Wrong windspeed unit for metric in open-meteo data for the weather plugin

Leberkasbepi, 2025-08-15:
https://github.com/fatihak/InkyPi/blob/a8b140a187335d501ee3b05123a6778ad17aedfa/src/plugins/weather/weather.py#L20
https://github.com/fatihak/InkyPi/blob/a8b140a187335d501ee3b05123a6778ad17aedfa/src/plugins/weather/weather.py#L37

Today I figured out, that the display show me always less windspeed then the open-meteo service. 
The reason is in codeline 20 and 37.
Open-Meteo give this unit for Europe in km/h, not in m/s. So the correct unit should be this.

`"speed": "km/h" `
` "metric":   "temperature_unit=celsius&wind_speed_unit=kmh&precipitation_unit=mm", `

k2v1n58, 2025-08-16:
Do you have any printscreen, mine is showing properly. Did you change something in the code?

[image]

Leberkasbepi, 2025-08-16:
The unit show it right, but the value was wrong. My API serve the unit in km/h.

Leberkasbepi, 2025-08-16:
Where did you get the example?
When I copy the[ API Url from the weather.py ](https://api.open-meteo.com/v1/forecast?latitude=47.07907668174417&longitude=15.411989092826843&hourly=temperature_2m,precipitation_probability,relative_humidity_2m,surface_pressure,visibility&daily=weathercode,temperature_2m_max,temperature_2m_min,sunrise,sunset&current_weather=true&timezone=auto&models=best_match&forecast_days=3)with my location, it show me km/h, but the code does not take that into account.

Leberkasbepi, 2025-08-17:
I tried the same thing and think I know what didn’t match. I only copied the URL and didn’t log it, so at my location it showed km/h. But the code originally specifies m/s.

So it’s fine.

**Sorry, my mistake!**

Where are the logs actually stored?