fatihak/InkyPi #291: Left and right boders missing on calendar on fresh OS and InkyPi install on waveshare epd7in3e display

Sovenger, 2025-08-26:
**Describe the bug**
The left and right CSS borders on the calendar are missing on a epd7in3e display.

**Expected behavior**
I'd expect the calendar to show left and right borders when viewed on a epd7in3e display.

**To Reproduce**
Steps to reproduce the behavior:
1. Install 64-bit lite version of raspy OS.
2. Install all dependencies for epd7in3e display from waveshare's website.
3. Install latest InkyPi from repo.
4. Create calendar on the Inky device and render it.
5. Observe that the calendar is missing the left and right borders, as if the screen isn't sized correctly at some point in the rendering pipeline.

**Logs**
```
Aug 26 16:07:56 calendar inkypi[560]: 16:07:56 - INFO - refresh_task - Manual update requested
Aug 26 16:07:56 calendar inkypi[560]: 16:07:56 - INFO - refresh_task - Refreshing plugin instance. | plugin_instance: 'Test calendar'
Aug 26 16:08:44 calendar inkypi[560]: 16:08:44 - INFO - refresh_task - Updating display. | refresh_info: {'refresh_type': 'Playlist', 'playlist': 'Default', 'plugin_id': 'calendar', 'plugin_instance': 'Test calendar', 'refresh_time': '2025-08-26T23:07:56.793834+00:00', 'image_hash': '459b9fdb83b9cd4b7b4724485ce2c2d6749a29bc37a6efaa11ab7d002cb89f24'}
Aug 26 16:08:44 calendar inkypi[560]: 16:08:44 - INFO - display.display_manager - Saving image to /home/pi/InkyPi/src/static/images/current_image.png
Aug 26 16:08:44 calendar inkypi[560]: 16:08:44 - INFO - display.waveshare_display - Displaying image to Waveshare display.
Aug 26 16:09:25 calendar inkypi[560]: 16:09:25 - INFO - display.waveshare_display - Putting Waveshare display into sleep mode for power saving.
Aug 26 16:11:16 calendar inkypi[560]: 16:11:16 - INFO - refresh_task - Manual update requested
Aug 26 16:11:16 calendar inkypi[560]: 16:11:16 - INFO - refresh_task - Refreshing plugin instance. | plugin_instance: 'Test calendar'
Aug 26 16:11:47 calendar inkypi[560]: 16:11:47 - INFO - refresh_task - Updating display. | refresh_info: {'refresh_type': 'Playlist', 'playlist': 'Default', 'plugin_id': 'calendar', 'plugin_instance': 'Test calendar', 'refresh_time': '2025-08-26T23:11:16.509477+00:00', 'image_hash': '1ade5dbd5dccca22cacea47c030207f69af9393347d3bdeef16026f1801055ea'}
Aug 26 16:11:47 calendar inkypi[560]: 16:11:47 - INFO - display.display_manager - Saving image to /home/pi/InkyPi/src/static/images/current_image.png
Aug 26 16:11:47 calendar inkypi[560]: 16:11:47 - INFO - display.waveshare_display - Displaying image to Waveshare display.
Aug 26 16:12:27 calendar inkypi[560]: 16:12:27 - INFO - display.waveshare_display - Putting Waveshare display into sleep mode for power saving.
```

**Screenshots**
[image]

**Additional context**
Raspberry Pi Model Zero 2 WH, Waveshare display epd7in3e 800x840.

Sovenger, 2025-08-27:
I'm assuming the borders should show since the YouTube video shows borders. If they have been removed since the video was recorded, what was the reasoning?

fatihak, 2025-08-29:
The borders were removed for visual purposes. I saw some calendar displays online that didn’t include them and I felt it looked cleaner.

fatihak, 2025-08-30:
Closing this since this intentional.

closed 2025-08-30 by fatihak