fatihak/InkyPi #292: Installation fails as PIP cannot download a package

PetarRusev86, 2025-08-27:
Hello all,
I am trying to install the InkyPi on a RPI W Zero, with latest rpi desktop image and I got error during the component installation:
Creating python virtual environment. 
        Installing python dependencies.  [-]   ERROR: HTTP error 403 while getting https://www.piwheels.org/simple/jiter/jiter-0.10.0-cp311-cp311-linux_armv6l.whl#sha256=82b8164820477f09f2d3f707125620519f88cf517afcd768cf1290b6030496e9 (from https://www.piwheels.org/simple/jiter/) (requires-python:>=3.9)
ERROR: Could not install requirement jiter<1,>=0.4.0 from https://www.piwheels.org/simple/jiter/jiter-0.10.0-cp311-cp311-linux_armv6l.whl#sha256=82b8164820477f09f2d3f707125620519f88cf517afcd768cf1290b6030496e9 (from openai==1.58.1->-r /home/madburbel/InkyPi/install/requirements.txt (line 9)) because of HTTP error 403 Client Error: Forbidden for url: https://www.piwheels.org/simple/jiter/jiter-0.10.0-cp311-cp311-linux_armv6l.whl for URL https://www.piwheels.org/simple/jiter/jiter-0.10.0-cp311-cp311-linux_armv6l.whl#sha256=82b8164820477f09f2d3f707125620519f88cf517afcd768cf1290b6030496e9 (from https://www.piwheels.org/simple/jiter/) (requires-python:>=3.9)

Python version is 3.11.2

Can you please advise how to proceed?

phattmatt, 2025-08-27:
Looks like the missing files are now available on the archive1.piwheels.org server.

I've just successfully installed all the prerequisites on a fresh Raspberry Pi OS 32-bit Lite instance.

If you are no longer seeing any errors with the PIP installs please consider closing this issue.

PetarRusev86, 2025-08-28:
Just tested, all working fine.
Issue was indeed in piwheels due to URL change: 
https://github.com/piwheels/packages/issues/544

closed 2025-08-28 by PetarRusev86