fatihak/InkyPi #310: Waveshare 7.3inch E Ink Spectra 6 (E6) stays blank and WebUI not accessable

spitzbirne32, 2025-09-18:
**Describe the bug**
A clear and concise description of what the bug is.
WebUI is not accessable after installing as well as display not turning on.

**Expected behavior**
WebUI being accessable via browser and display turned on.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash latest raspbian lite 64 bit to sd card for pi4 or pi zero 2 (both have the same outcome on my end)
2. install git
3. run install commands for 7.3inch E Ink Spectra 6 (E6)  --> sudo bash install/install.sh -W epd7in3e
4. Enter Y when asked to reboot

**Logs**
Provide the InkyPi service logs, see [troubleshooting.md](./docs/troubleshooting.md) for instructions on how to view logs.
`pi@inkypi:~ $ sudo systemctl status inkypi
● inkypi.service - InkyPi App
     Loaded: loaded (/etc/systemd/system/inkypi.service; enabled; preset: enabled)
     Active: active (running) since Thu 2025-09-18 23:24:03 CEST; 6min ago
   Main PID: 656 (bash)
      Tasks: 9 (limit: 3921)
        CPU: 16.624s
     CGroup: /system.slice/inkypi.service
             ├─656 bash /usr/local/bin/inkypi
             └─681 python -u /home/pi/InkyPi/src/inkypi.py
Sep 18 23:24:03 inkypi systemd[1]: Started inkypi.service - InkyPi App.
Sep 18 23:24:06 inkypi inkypi[681]: /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/inky/__init__.py:18: UserWarning: pkg_resources is deprecated>
Sep 18 23:24:06 inkypi inkypi[681]:   from pkg_resources import declare_namespace
Sep 18 23:24:06 inkypi inkypi[681]: 23:24:06 - INFO - __main__ - Starting InkyPi in PRODUCTION mode on port 80
Sep 18 23:24:06 inkypi inkypi[681]: 23:24:06 - INFO - display.waveshare_display - Initializing Waveshare display
Sep 18 23:24:06 inkypi inkypi[681]: 23:24:06 - INFO - display.waveshare_display - Loading EPD display for epd7in3e display`
After restarting the service:
`pi@inkypi:~ $ sudo systemctl restart inkypi.service
pi@inkypi:~ $ journalctl -u inkypi -n 100
Sep 18 23:24:03 inkypi systemd[1]: Started inkypi.service - InkyPi App.
Sep 18 23:24:06 inkypi inkypi[681]: /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/inky/__init__.py:18: UserWarning: pkg_resources is deprecated>
Sep 18 23:24:06 inkypi inkypi[681]:   from pkg_resources import declare_namespace
Sep 18 23:24:06 inkypi inkypi[681]: 23:24:06 - INFO - __main__ - Starting InkyPi in PRODUCTION mode on port 80
Sep 18 23:24:06 inkypi inkypi[681]: 23:24:06 - INFO - display.waveshare_display - Initializing Waveshare display
Sep 18 23:24:06 inkypi inkypi[681]: 23:24:06 - INFO - display.waveshare_display - Loading EPD display for epd7in3e display
Sep 18 23:34:03 inkypi inkypi[681]: Traceback (most recent call last):
Sep 18 23:34:03 inkypi inkypi[681]:   File "/home/pi/InkyPi/src/inkypi.py", line 60, in <module>
Sep 18 23:34:03 inkypi inkypi[681]:     display_manager = DisplayManager(device_config)
Sep 18 23:34:03 inkypi inkypi[681]:                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Sep 18 23:34:03 inkypi inkypi[681]:   File "/home/pi/InkyPi/src/display/display_manager.py", line 53, in __init__
Sep 18 23:34:03 inkypi inkypi[681]:     self.display = WaveshareDisplay(device_config)
Sep 18 23:34:03 inkypi inkypi[681]:                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Sep 18 23:34:03 inkypi inkypi[681]:   File "/home/pi/InkyPi/src/display/abstract_display.py", line 19, in __init__
Sep 18 23:34:03 inkypi inkypi[681]:     self.initialize_display()
Sep 18 23:34:03 inkypi inkypi[681]:   File "/home/pi/InkyPi/src/display/waveshare_display.py", line 64, in initialize_display
Sep 18 23:34:03 inkypi inkypi[681]:     self.epd_display_init()
Sep 18 23:34:03 inkypi inkypi[681]:   File "/home/pi/InkyPi/src/display/waveshare_epd/epd7in3e.py", line 112, in init
Sep 18 23:34:03 inkypi systemd[1]: Stopping inkypi.service - InkyPi App...
Sep 18 23:34:03 inkypi inkypi[681]:     self.ReadBusyH()
Sep 18 23:34:03 inkypi inkypi[681]:   File "/home/pi/InkyPi/src/display/waveshare_epd/epd7in3e.py", line 92, in ReadBusyH
Sep 18 23:34:03 inkypi inkypi[681]:     epdconfig.delay_ms(5)
Sep 18 23:34:03 inkypi inkypi[681]:   File "/home/pi/InkyPi/src/display/waveshare_epd/epdconfig.py", line 99, in delay_ms
Sep 18 23:34:03 inkypi inkypi[681]:     time.sleep(delaytime / 1000.0)
Sep 18 23:34:03 inkypi inkypi[681]: KeyboardInterrupt
Sep 18 23:34:04 inkypi systemd[1]: inkypi.service: Deactivated successfully.
Sep 18 23:34:04 inkypi systemd[1]: Stopped inkypi.service - InkyPi App.
Sep 18 23:34:04 inkypi systemd[1]: inkypi.service: Consumed 28.709s CPU time.
Sep 18 23:34:04 inkypi systemd[1]: Started inkypi.service - InkyPi App.
Sep 18 23:34:06 inkypi inkypi[758]: /usr/local/inkypi/venv_inkypi/lib/python3.11/site-packages/inky/__init__.py:18: UserWarning: pkg_resources is deprecated>
Sep 18 23:34:06 inkypi inkypi[758]:   from pkg_resources import declare_namespace
Sep 18 23:34:07 inkypi inkypi[758]: 23:34:07 - INFO - __main__ - Starting InkyPi in PRODUCTION mode on port 80
Sep 18 23:34:07 inkypi inkypi[758]: 23:34:07 - INFO - display.waveshare_display - Initializing Waveshare display
Sep 18 23:34:07 inkypi inkypi[758]: 23:34:07 - INFO - display.waveshare_display - Loading EPD display for epd7in3e display`
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Provide the Raspberry Pi Model, Display Model & Size, and any other context.
Raspberry Pi 4B 4GB, 32GB SD Card
Raspberry Pi Zero 2 WH, 32GB SD Card

I did check https://github.com/fatihak/InkyPi/blob/main/docs/troubleshooting.md but nothing fits my case.
The only thing that might fit is the "Waveshare e-Paper EPD Devices" section, but I have no idea how to do this without possibly breaking something.

Any help is appreciated. Thanks.

F2011, 2025-10-04:
Did you check whether your Display is working?
Try [this waveshare test program (link leads to the documentation)](https://github.com/waveshareteam/e-Paper/blob/master/RaspberryPi_JetsonNano/python/readme_rpi_EN.txt) to check if your display is working. If the console output of the test program is something like `e-Paper busy H`, your display is not working properly.

gianlucapisati, 2025-10-28:
Have you installed as well `liblgpio1` with `sudo apt-get install -y liblgpio1`?

then try again with 

```
sudo /usr/local/inkypi/venv_inkypi/bin/pip install --no-cache-dir gpiozero==2.0.1 colorzero==2.0 spidev
sudo /usr/local/inkypi/venv_inkypi/bin/pip install --no-cache-dir lgpio
```

Then 

```
sudo /usr/local/inkypi/venv_inkypi/bin/python - <<'PY'
import gpiozero
print("gpiozero import OK")
import lgpio
print("lgpio import OK")
PY
```

should print `lgpio import OK` as well.

the environment variables

```
Environment=GPIOZERO_PIN_FACTORY=lgpio
Environment=WAVESHARE_DEVICE=epd7in3e
```

needs to be added in the `[Service]` section before `ExecStart=/usr/local/bin/inkypi`

phattmatt, 2025-11-02:
> also encountered the lgpio wheel error on a fresh install on Trixie.
> 
> what helped me was to manually install lg using these steps:
> 
> ```
> 1. Install build tools and prerequisites
> sudo apt update
> sudo apt install git build-essential python3-dev
> 2. 
> 3. Clone the official lg repositorycd ~/
> git clone https://github.com/joan2937/lg.git
> cd lg
> 
> 3. Build and install the C library lg
> make
> sudo make install
> 
> 4. Verify installation
> ls /usr/local/lib | grep lgpio
> ```
> 
> then run the InkyPi install script again. however this only aid in installing without errors. the E6 still stays blank and there's no Web service/UI running/accessible

If you want to avoid the manual install then the required library is also available as a system package: `liblgpio-dev`.

nielsvdp, 2026-01-06:
I also have a Waveshare 7.3inch Spectra 6 (epd7in3e) and Pi Zero 2 W running Trixie. The InkyPi service installs and starts, but I get no image. The Waveshare testing program does work and shows images, InkyPi logs show no error except for a warning.

logs:

```
Jan 06 23:04:35 PiFrame systemd[1]: Started inkypi.service - InkyPi App.
Jan 06 23:04:57 PiFrame inkypi[1016]: /usr/local/inkypi/venv_inkypi/lib/python3.13/site-packages/inky/__init__.py:18: UserWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html. The pkg_resources package is slated for removal as early as 2025-11-30. Refrain from using this package or pin to Setuptools<81.
Jan 06 23:04:57 PiFrame inkypi[1016]:   from pkg_resources import declare_namespace
Jan 06 23:04:58 PiFrame inkypi[1016]: 23:04:58 - INFO - __main__ - Starting InkyPi in PRODUCTION mode on port 80
Jan 06 23:04:58 PiFrame inkypi[1016]: 23:04:58 - INFO - display.waveshare_display - Initializing Waveshare display
Jan 06 23:04:58 PiFrame inkypi[1016]: 23:04:58 - INFO - display.waveshare_display - Loading EPD display for epd7in3e display
Jan 06 23:05:10 PiFrame inkypi[1016]: 23:05:10 - INFO - refresh_task - Starting refresh task
Jan 06 23:05:10 PiFrame inkypi[1016]: 23:05:10 - INFO - waitress - Serving on http://0.0.0.0:80
```

Is it possible the warning about pkg_resources is causing this?