fatihak/InkyPi #314: piwheels fails - certificate expired

ps-gregg, 2025-09-25:
Running install and it looks like piwheels certificate is expired -  install fails 

$ sudo bash install/install.sh
Checking if inkypi.service is running
        inkypi.service not running [✔]
Enabling interfaces required for inkypi
        SPI Interface has been enabled. [✔]
        I2C Interface has been enabled. [✔]
Enabling single CS line for SPI interface in config.txt
Fetch available system dependencies updates.  [|] W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Fetch available system dependencies updates.  [✔]
Installing system dependencies.  [✔]
Enabling and starting zramswap service.
Enabling and starting earlyoom service.
Synchronizing state of earlyoom.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable earlyoom
Installing inkypi to /usr/local/inkypi
        Creating symlink from /home/pinky/InkyPi/install/../src to /usr/local/inkypi/src [✔]
Creating python virtual environment.
  WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/setuptools/setuptools-80.9.0-py3-none-any.whl
        Installing python dependencies.  [-]   WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:992)'))': /simple/inky/inky-2.2.1-py3-none-any.whl
        Installing python dependencies.  [-]   WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:992)'))': /simple/inky/inky-2.2.1-py3-none-any.whl
        Installing python dependencies.  [/]   WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:992)'))': /simple/inky/inky-2.2.1-py3-none-any.whl
        Installing python dependencies.  [\]   WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:992)'))': /simple/inky/inky-2.2.1-py3-none-any.whl
        Installing python dependencies.  [/]   WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:992)'))': /simple/inky/inky-2.2.1-py3-none-any.whl
        Installing python dependencies.  [-] ERROR: Could not install packages due to an OSError: HTTPSConnectionPool(host='archive1.piwheels.org', port=443): Max retries exceeded with url: /simple/inky/inky-2.2.1-py3-none-any.whl (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:992)')))

Raspberry Pi Model 2W
Display Inky Impression 7.3 Spectra

ps-gregg, 2025-09-25:
# Workaround:

Modified pip.conf by typing

```
~/InkyPi $ sudo nano /etc/pip.conf
```

[global]
# extra-index-url=https://www.piwheels.org/simple   <comment this line>

save changes and then run 

```
~/InkyPi $  SCRIPT_DIR=$( cd -P "$( dirname "$SOURCE" )" >/dev/null 2>&1 && pwd )
APPNAME="inkypi"
INSTALL_PATH="/usr/local/$APPNAME"
SRC_PATH="$SCRIPT_DIR/../src"
BINPATH="/usr/local/bin"
VENV_PATH="$INSTALL_PATH/venv_$APPNAME"

SERVICE_FILE="$APPNAME.service"
SERVICE_FILE_SOURCE="$SCRIPT_DIR/$SERVICE_FILE"
SERVICE_FILE_TARGET="/etc/systemd/system/$SERVICE_FILE"

APT_REQUIREMENTS_FILE="$SCRIPT_DIR/debian-requirements.txt"
PIP_REQUIREMENTS_FILE="$SCRIPT_DIR/install/requirements.txt"
```

and then run 

```
~/InkyPi $   sudo 
$VENV_PATH/bin/python -m pip install --no-cache-dir --index-url https://pypi.org/simple -r $PIP_REQUIREMENTS_FILE
```

wait 30-40 minutes........... and received error compiling cysystemd.   run

```
~/InkyPi $  sudo apt update
~/InkyPi $  sudo apt install libsystemd-dev
```

and then run 

```
~/InkyPi $  sudo $VENV_PATH/bin/python -m pip install --no-cache-dir --index-url https://pypi.org/simple -r $PIP_REQUIREMENTS_FILE
```
wait 30-40 minutes.......... again and run

```  
~/InkyPi $   sudo reboot now
```

and all is working....

phattmatt, 2025-09-29:
Thanks for the follow up.

Looks like the certificate for "archive1.piwheels.org" was renewed around the time you saw this issue, with the current one reporting:

```
Valid from: Thu, 25 Sep 2025 10:29:43 UTC
Valid until: Wed, 24 Dec 2025 10:29:42 UTC
```

Closing as resolved, since it seems to be working now, and is not an issue with InkyPi.

closed 2025-09-29 by phattmatt